john-/ham2mon #5: Classify audio via a pre-trained model

john-, 2023-11-01:
This enhancement will classify audio recordings to distinguish between voice, data and those that are to be skipped (open mic, static, etc).

john-, 2023-11-03:
https://github.com/john-/ham2mon/tree/classify_recording_squashed

john-, 2023-11-03:
This release contains the tflite mode required for this branch:

[v0.1-alpha](https://github.com/john-/ham2mon/releases/tag/v0.1-alpha)

It can be placed in the `apps/model` directory in the cloned repository or from the release's zip file that was unzipped.

lordmorgul, 2023-12-16:
I have been testing the default trained model against a GMRS repeater that has frequent morse ID and the model has done well rejecting those morse audio sequences.  It still captures some static only but much less often than without having the voice classification turned on.

john-, 2023-12-16:
Good to hear.

Interesting that you mention the morse audio sequences.  I have seem some cases where these were detected as voice.  This is probably because the data I trained with didn't include much of this.  At some point I will do another training run with some more of these included.  If you find there is some improvement needed here feel free to send me some examples and I should be able to include them in the training.

Static is more tricky.  The "skip" category is pretty subjective the way I approached it.  Maybe there is a better way.  Basically, anything I didn't want to listen to that wasn't data was classified as "skip".  This was subjective in that there are cases where a person talking with "too much" static I labeled as "skip".  It was me that determined what "too much" meant and not an algorithm.

Again there mat be a better approach with this and I am open to ideas.

lordmorgul, 2024-02-04:
Reviewed the classify audio commits up to master today and have no findings, and have been test running at master for weeks.

john-, 2024-02-26:
This is now merged so closing.

closed 2024-02-26 by john-