john-/ham2mon #15: Treatment of "demodulator.center_freq == 0"

john-, 2024-02-29:
I have noticed that if detected center frequency is "0" (zero) then this line stops it from being added to a demodulator:

https://github.com/john-/ham2mon/blob/4e33653820ac1a9f144d00ff0b305ee34cd1aa70/apps/scanner.py#L302

A channel of "0" won't be added initially because `self.receiver.get_demod_freqs()` contains a "0" so the center_freq of "0" exists.  It won't be added after all demodulators have frequencies assigned because none of the demodualtion frequencies are "0".

The code has been this way for years.

Am I missing something?  Is this intentional to avoid dc spike on those receivers that have it?

john-, 2024-02-29:
This is intentional....from the README:

The demodulators are parked at 0 Hz baseband when not tuned, as this provides a constant, low amplitude signal due to FM demod of LO leakage.

lordmorgul, 2024-03-04:
I think that there is no easy way to identify valid signal at 0Hz, at least before without classification.  It is possible that class id could be used to do that but would be cpu cycles classifying nothing all the time.

john-, 2024-03-04:
Thanks for the input.  I just left the code the way it was.

Closing this ticket.

closed 2024-03-04 by john-