john-/ham2mon #16: Upcoming change: Scanner:scan_cyccle reorganization

john-, 2024-03-01:
Over time the current code in scan_cycle has grown and been modified such that it is not clear immediately what is happening.  Also, it provides the GUI separate lists that contain active frequencies and tuned frequencies.

This change will re-organize the function and a single frequency list that contains metadata for each channel.

lordmorgul, 2024-03-04:
GUI could either display Hz or keep MHz and have code all ref same list but with the multiplies in every instance of referencing the list.  That is clearer code but slightly more operations.  Or the list could contain all three: baseband, RF Hz and RF MHz frequencies in it per entry, making more memory but much clearer what is happening.

john-, 2024-03-04:
Here are some performance timings.  As ham2mon needs to be manually stopped after replay completes that will create some variance.  All timings are with ham2mon windows active and visible.
```
branch: scan_cycle_reorg_squashed
260.82 user+sys | 198.01s user 62.81s system 225% cpu 1:55.77 total
264.39 user+sys | 201.29s user 63.10s system 226% cpu 1:56.72 total
263.88 user+sys | 201.75s user 62.13s system 226% cpu 1:56.34 total

branch: master
263.32 user+sys | 200.88s user 62.44s system 226% cpu 1:56.43 total
263.25 user+sys | 197.35s user 65.90s system 225% cpu 1:56.52 total
262.99 user+sys | 196.83s user 66.16s system 226% cpu 1:56.05 total
```

Based on this it does seems the performance is comparable between this branch and master.  At least nothing has broken with the changes.

closed 2024-03-11 by john-