john-/ham2mon #18: Channel Log Reorg

john-, 2024-03-10:
After some [discussion](https://github.com/lordmorgul/ham2mon/discussions/19) I have done some work on channel log function.  This is in a separate [branch](https://github.com/john-/ham2mon/tree/channel-log_reorg_squashed):

## User impacting changes
1. Command line options have changed:
```
  -T CHANNEL_LOG_TYPE, --log_type CHANNEL_LOG_TYPE
                        Log file type for channel detection
  -L CHANNEL_LOG_TARGET, --log_target CHANNEL_LOG_TARGET
                        Log file or endpoint for channel detection
  -A CHANNEL_LOG_TIMEOUT, --log_active_timeout CHANNEL_LOG_TIMEOUT
                        Timeout delay for active channel log entries
```
2. Th GUI shows only the log_type and log_target.
3. If no options are given no channel logging will be performed
In this case the gui will not show log_type or log_target

The option `--log-type fixed-field` generally shows the same information as from before this change.  Any gaps should be able to be addressed with this revised approach,

## Backend changes
- All logic moves from Scanner into Receiver.

## New capability
See the [README](https://github.com/john-/ham2mon/tree/channel-log_reorg_squashed?tab=readme-ov-file#channel-detection-log-file) for details.

john-, 2024-03-11:
There is a bug I am trying to track down with this.  The channel-log_reorg_squashed branch is not yet working.

john-, 2024-03-11:
The issue has been addressed.

lordmorgul, 2024-03-17:
Channel log appears to be working fine here. 
I looked back at old format reason and was intending that it track amplitude of the peak for the channel center spectrum bin in the log; this was the column hard coded 0 dB.  The other column coded 0 dB was supposed to be the threshold squelch level at the time.
Not having any need for that data right now it makes sense to leave those out and see how this logging works over awhile.
I recommend merging this to master.

lordmorgul, 2024-03-17:
For future note the time field could probably be reduced, but in a database time field it is more optimized to just use a datetime than a smaller field so it makes sense to keep this extra for consistency.  If one had a large file of channel logs and wanted to transfer it to a database format for instance the extra time field accuracy would be better to have than not.

john-, 2024-03-17:
Thanks for looking at this.

Yeah the time is pretty verbose.  For that and the other items you mentioned they can be changed down the road.

lordmorgul, 2024-03-17:
Yes later on.  The demod type used, and foxed frequency accuracy/digits would be good adds for later also.

john-, 2024-03-19:
This has been merged into master.

closed 2024-03-19 by john-