john-/ham2mon #22: Reorg: Demodulator classes into seperate files

john-, 2024-03-16:
In preparation for [new modulators](https://github.com/john-/ham2mon/issues/8#issuecomment-2000606820) it makes sense to separate out the demodulators into separate files.  This issue is to track approach and changes in this area.

john-, 2024-03-16:
From what you have seen is the the case that new demodulators (beyond NBFM and AM) work with current Receiver class logic or have you had to modify it with anything demodulator specific?  For that matter has there been need to change the BaseTuner class?

lordmorgul, 2024-03-17:
So far wbfm demodulator and adding CTCSS filtering were both done in the specific demodulator not the BaseTuner class.  I do not think I had to edit the base at all.
I expect that SSB should be the same when added.

The Receiver class was changed to add the option of each demod type, but that is all.  It would be possible to do bin sample sizing adjustments per demod type but I did not do that and its probably not worth it really.  As long as we do not add a demod that needs to have smaller bins than NBFM should be ok.  TBD if the SSB demod will need that I have not considered it.

https://github.com/lordmorgul/ham2mon/blob/c205cbda7055fca46ac46ef0285e7a75ad34aaea/apps/receiver.py#L846C1-L859C88

john-, 2024-03-17:
Good to hear on impact.  Based on that I will take a look at this change and maybe first pass at bringing in the WBFM demodulator.

john-, 2024-03-18:
Here is the [commit](https://github.com/john-/ham2mon/commit/5560ab8ce45a5a7c4e30af0330cb8a2dba6da354) that brings in the refactored demodulator files.  It is also in the [wbfm branch](https://github.com/john-/ham2mon/tree/wbfm_squashed).

lordmorgul, 2024-03-23:
Initial testing shows NBFM is working in that branch where separated.  It works with and without classification on.

AM exclusion tested (AM demod but FM signal), it did save a file when FM signal was an active channel but did not record any audio (expected result, though not saving a blank file would be better I think this has been the behavior all along).  Will test an AM signal still works later.

WBFM works with no classification turned on.
WBFM is not saving files when voice classification is on; 16bit is on.  I suspect the audio sampled appears different enough in audio spectrum that it is not matching the training.
This WBFM_squashed branch has a mic of the ctcss code in the WBFM demod but not in the NBFM demod.

Neither of the demods are filtering out the tone from my repeater which leaves a loud overpowering tone in the recording.  The ctcss code in WBFM was supposed to filter that out of the audio before it gets to speaker or saved file, so I need to look into why before copying that code into NBFM.

john-, 2024-03-23:
Regarding this:

>Neither of the demods are filtering out the tone from my repeater which leaves a loud overpowering tone in the recording. The ctcss code in WBFM was supposed to filter that out of the audio before it gets to speaker or saved file, so I need to look into why before copying that code into NBFM.

The WBFM branch was intended for the WBFM demod only.  From the way I see the commit you linked above the ctcss code is either commented out or disabled due to defaults in the WBFM demodulator init.

Also, I did not bring over any changes to the NBFM demodulator so that one doesn't even have the commented out ctcss code.

Agreed it makes sense to get things working in the WBFM code before bringing into the NBFM demodulator.

john-, 2024-03-23:
Regarding this:

>WBFM works with no classification turned on.
WBFM is not saving files when voice classification is on; 16bit is on. I suspect the audio sampled appears different enough in audio spectrum that it is not matching the training.
This WBFM_squashed branch has a mix of the ctcss code in the WBFM demod but not in the NBFM demod.

Good to hear that the basics of WBFM are working.

I will still try and get a WBFM signal I can test with.

Regarding classification:  It would be good to isolate the problem.  One way is to run the classifier on a specific file or two.  The classifier.py has a main that will test classification with a audio file on disk.  You can add a test case(s) for wbfm files.  Like this:
```
print('should be voice (V): ' + classifier.is_wanted("test/voice_wbfm.wav"))
```

Put a sample known voice file that is at least a second long named with the name you used into the test directory.

john-, 2024-03-23:
Regarding this:

>AM exclusion tested (AM demod but FM signal), it did save a file when FM signal was an active channel but did not record any audio (expected result, though not saving a blank file would be better I think this has been the behavior all along). Will test an AM signal still works later.

I am not sure either what the expected results should be.  Let me know how it goes with your testing.

lordmorgul, 2024-03-31:
I missed that comment on trying to classifier directly, I have saved a few files with misses that I will try with that.

john-, 2024-03-31:
> I missed that comment on trying to classifier directly, I have saved a few files with misses that I will try with that.

Also note that ham2mon will start without error if you specify classification (e.g. --voice) without a model where it expects.  The --debug option will indicate the failure to load the model but the UI will give impression classification is happening.  It will record without classifying.

Sounds like this may not be your problem but worth a check.

john-, 2024-05-21:
This is now merged into master.  The CTCSS changes are commented out and untested.  This needs to be addressed in a separate ticket.

closed 2024-05-28 by john-