john-/ham2mon #23: WBFM integration

john-, 2024-03-18:
Here is a first pass at [WBFM integration](https://github.com/john-/ham2mon/tree/wbfm_squashed).  This also includes splitting out demodulators into their own files.  That is a separate commit if it needs to be isolated.

I am still trying to track down a frequency to test WBFM it but there are no errors and audio files are generated (tested only with recording).

lordmorgul, 2024-03-30:
Confirming when you said “ tested only with recording” you mean without classification? And that recording was WBFM modulator with NBFM signal, and does save audio.  I confirmed that does work.  WBFM signal with WBFM modulator also worked.  WBFM with classification does not work but likely is just due to re-training need.  I will backlog collect signals for a training set.

john-, 2024-04-02:
> Confirming when you said “ tested only with recording” you mean without classification? 

Yes, I did not test classification.  I only confirmed that with the -w options files were written to disk.  This was not with a WBFM signal as I think ham2mon will record transmissions even if they are not of the requested type.

john-, 2024-05-21:
This is now merged into master.  Based on this thread there may still be open issue with audio classification of WBFM recordings.

closed 2024-05-28 by john-