john-/ham2mon #26: Auto Priority

john-, 2024-05-17:
If audio classification is enabled and channel has mostly voice, add that channel to the priority list.

I have the basics of this working and will be testing it further.

Questions:

- There probably needs to be an option to disable this?  For me that doesn't seem to be the case yet.
-  Current logic makes voice transmissions a priority.  Should there be an option to make data transmissions a priority?  Again, this would not apply to be and not sure if anyone would be interested in it.

john-, 2024-05-20:
Basics working here:

https://github.com/john-/ham2mon/tree/auto_priority

lordmorgul, 2024-05-21:
Great idea here.  I think generally data tends to have more defined channels, at least for use cases I would have to record it there would likely be known frequencies (WINLINK or APRS, etc).  I wouldn't use this for data and instead assign priority from file.
For voice this sounds great though.

john-, 2024-05-21:
Thanks for the insight.

So far it seems to do a pretty good job.  Initially I was thinking I would get more voice recordings.  However, it turns out it is about the same number recorded from the the test file I replay.  What happens is the recordings are longer.  I compared by file size and it was over twice the amount of audio recorded with this option enabled.

john-, 2024-05-21:
This is now merged into master

closed 2024-05-28 by john-